TAMCTec/gt911-arduino #2: New To Allow 2nd I2C Bus

TryingToGITthis, 2022-06-03:
Need a modification where a 2nd I2C bus can be used and not just the default Wire.

wBrhy2, 2023-06-30:
Isn't this enabled by the fact that you have to declare what the pin mapping is?

TAMCTec, 2023-07-27:
> Need a modification where a 2nd I2C bus can be used and not just the default Wire.

Try just use another SDA and SCL pin

OlliHenri, 2024-07-16:
As I understand, it is not enough just to use other SDA and SCL pins.
To use the two I2C bus interfaces of the ESP32, you need to create two TwoWire instances.
There are CYD display boards with GT911 that need to declare 2 i2c busses to use.
So it would be good like ADAFRUIT does it in many libs, to offer a way to hand a dedicated i2c instance to the GT911 library.