TAMCTec/gt911-arduino #7: GT911 stops after reset ESP32-8048S070 Sunton store

HowardAtHome, 2023-03-25:
After much frustration I have finally worked out the problem and solution.

The GT911 really needs the INT pin connected, this is used to set the I2C address of the chip on bootup, then can be used to drive interrupt driven touch. On my board this is not connected and floats, hence randomly changes between the 2 addresses on bootup.

Easy to sort in the TAMC_GT911.cpp. In the reset () add
    delay(50);
    **checkCorrectAddr ();**
    readBlockData(configBuf, GT911_CONFIG_START, GT911_CONFIG_SIZE);

Then create a new routine

void TAMC_GT911::checkCorrectAddr(void) {
//Need to solve floating INT pin so on reset changes between the 2 addresses
  uint8_t returnSize;
  Wire.beginTransmission(addr);
  Wire.write(highByte(GT911_PRODUCT_ID));
  Wire.write(lowByte(GT911_PRODUCT_ID));
  Wire.endTransmission();
  returnSize = Wire.requestFrom(addr, (uint8_t)1); //returns rxLength - if 0 we have a problem
  if (returnSize == 0) {
    addr = GT911_ADDR2;
    Serial.println ("setting address to ADDR2");
  }
}

hallard, 2023-04-01:
you saved my I suspected something like that with randomly working touch screen. Would you be able to do a PR to fix?

I can do that be need to be sure @TAMCTec will allow the PR to be merged with increased version release?

hallard, 2023-04-01:
also you can avoid changing this library code just adding after ts.init() in your main code 

```cpp
  // Begin init wire, and Reset the device also, no need wire.begin() to be done here
  //Wire.begin(TOUCH_GT911_SDA, TOUCH_GT911_SCL);
  ts.begin(GT911_ADDR1);

  // Need to solve floating INT pin so on reset changes between the 2 addresses
  Wire.beginTransmission(GT911_ADDR1);
  Wire.write(highByte(GT911_PRODUCT_ID));
  Wire.write(lowByte(GT911_PRODUCT_ID));
  Wire.endTransmission();
  //returns rxLength - if 0 we have a problem
  uint8_t returnSize = Wire.requestFrom(GT911_ADDR1, (uint8_t)1); 
  if (returnSize == 0) {
    Serial.println("Setting address to ADDR2");
    // restart with other address
    ts.begin(GT911_ADDR2);
  }

   // Now do your normal job
  ts.setRotation(TOUCH_GT911_ROTATION);
```

eagl1, 2024-10-17:
> checkCorrectAddr

When to call this function ?

TractorHound, 2025-02-04:
Further playing around with this and there are still issues with my ESP32-3248035C board with configuration values apparently not writing consistently. However I found two useful mods:
1) The reset circuit uses a 0.1uF capacitor between RST and ground. Espressif however recomends a value between 1 and 10uF. Changing to the larger value (I used 10uF) and it isn't necessary to press BOOT when downloading from the Arduino IDE as it works automatically and then automatically restarts after loading.
2) Resistor location R25 isn't populated. Inserting a zero ohm resistor or shorting link here joins IO21 to the GT911's INT pin. The touch screen then works without any problems using the standard TAMC_GT911 library.

GrantGeno, 2025-02-14:
Glad to have found this.  More than a year later I see that both the 7" and 5" inch versions of this board still have the GT911 INT pin (AKA TP_INT) unconnected. On the current revision of the board, the INT pin goes to R17 which is not populated. The other side of R17 goes to IO18, but we all know that the circuit is short of pins, so using IO18. Nearby are the I2C pullups giving access to 3.3v. 

**I wonder if anyone has tried using a pullup or pulldown instead of just leaving it floating?**  I think the board could be fitted with a resistor to either 3v3 or GND nearby.

GrantGeno, 2025-02-14:
I found some other 5" and 7"boards, and it looks like more than a few companies make these, that has a little circuit on the digitizer FPC connected to GT911 pin 16 (INT). Then on the main board, R17 is not populated, as I described above, but looking closely at the connector FPC1, you can see that the INT pin is GROUNDED.  That ground connection is not shown on the schematic anywhere!

eagl1, 2025-02-14:
The INT pin is the cause of the problem, I had to cut its line on the board and it worked.

My display is 7" EA7013:

[image]