TAMCTec/gt911-arduino #10: Issue using the library with a 3.5"ILI9488/GT911 capacitive touchscreen, power by an ESP32 devmodule

BMWK1, 2023-11-04:
Trying to use a 3.5" TFT capacitive touchscreen, ILI9488 driver, GT911 I2C bus for touch commands, with an ESP32 Dev module.

i have read the previous issues, none corrects my problem.
Display works with TFT_eSPI library from Bodmer
I2C bus works, i included a test code which returns me a valid address (0x14)
I run the example code, with the pin setting for an ESP32 (SDA 21 , SCL 22), and it returns always the same looping values (every **15** seconds), without any screen touch, and touching the screen doesn't produce any difference.
I have uncommented some "println" in the .cpp file, so the 6 extra lines after each Touch loop.
Touch 1:  x: 65535 y: 65535 size: 255
Touch 2:  x: 65535 y: 65535 size: 255
Touch 3:  x: 65535 y: 65535 size: 255
Touch 4:  x: 65535 y: 65535 size: 255
Touch 5:  x: 65535 y: 65535 size: 255
Touch 6:  x: 65535 y: 65535 size: 255
Touch 7:  x: 65535 y: 65535 size: 255
Touch 8:  x: 65535 y: 65535 size: 255
Touch 9:  x: 65535 y: 65535 size: 255
Touch 10:  x: 65535 y: 65535 size: 255
Touch 11:  x: 65535 y: 65535 size: 255
Touch 12:  x: 65535 y: 65535 size: 255
Touch 13:  x: 65535 y: 65535 size: 255
Touch 14:  x: 65535 y: 65535 size: 255
Touch 15:  x: 65535 y: 65535 size: 255
TAMC_GT911::read
bufferStatus: 1
largeDetect: 1
proximityValid: 1
haveKey: 1
touches: 15

except the first occurence after reset, where the values are
Touch 1:  x: 321 y: 481 size: 255
Touch 2:  x: 321 y: 481 size: 255
Touch 3:  x: 321 y: 481 size: 255
Touch 4:  x: 321 y: 481 size: 255
Touch 5:  x: 321 y: 481 size: 255
Touch 6:  x: 321 y: 481 size: 255
Touch 7:  x: 65535 y: 65535 size: 255
Touch 8:  x: 65535 y: 65535 size: 255
Touch 9:  x: 65535 y: 65535 size: 255
Touch 10:  x: 65535 y: 65535 size: 255
Touch 11:  x: 65535 y: 65535 size: 255
Touch 12:  x: 65535 y: 65535 size: 255
Touch 13:  x: 65535 y: 65535 size: 255
Touch 14:  x: 65535 y: 65535 size: 255
Touch 15:  x: 65535 y: 65535 size: 255

Any help to understand the issue ?

the test code
#include "TAMC_GT911.h"

#define TOUCH_SDA 21
#define TOUCH_SCL 22
#define TOUCH_INT 25
#define TOUCH_RST 13
#define TOUCH_WIDTH 320
#define TOUCH_HEIGHT 480

TAMC_GT911 tp = TAMC_GT911(TOUCH_SDA, TOUCH_SCL, TOUCH_INT, TOUCH_RST, TOUCH_WIDTH, TOUCH_HEIGHT);

void setup() {
  Serial.begin(115200);
  Serial.println("TAMC_GT911 Example: Ready");
  tp.begin();
  tp.setRotation(ROTATION_NORMAL);
}

void loop() {
  tp.read();
  if (tp.isTouched){
    for (int i=0; i<tp.touches; i++){
      Serial.print("Touch ");Serial.print(i+1);Serial.print(": ");;
      Serial.print("  x: ");Serial.print(tp.points[i].x);
      Serial.print("  y: ");Serial.print(tp.points[i].y);
      Serial.print("  size: ");Serial.println(tp.points[i].size);
      Serial.println(' ');
    }
  }
}

BMWK1, 2023-12-03:
It makes sense, i'm going to test with different adresses.

eagl1, 2024-10-17:
I can't change the resolution to 800x480.